doctor64/tuyaZigbee #1: Hi, is there work still ongoing for TS0505B 5 channel RGBCW bulb?

baudneo, 2023-09-16:
I am looking to replace tuya firmware on 3 TS0505B bulbs. I am uncertain if the bulbs would be supported?

Thank you for your work!

doctor64, 2023-10-01:
Hello, sorry for late reply.
Unfortunately, i found what TSXXXX indexes from Tuya (like TS0505, etc) are unreliable. It's more like device class, but under same index can be different devices, even build on different zigbee chips.
So, i cannot guarantee what my firmware, which developed for LED controller, will be operational on your LED bulb.
Can you please provide zigbee MAC address of device, to verify it based on Telink chip, and link to Aliexpress or other store where I can get device for testing?

baudneo, 2023-10-01:
Link: https://www.aliexpress.com/item/1005004093785752.html

MAC: 0xa4c13853f145e1da

doctor64, 2023-10-01:
Ok, MAC starts from 0xA4C138, so it's Telink.
I ordered lamp, but keep in mind delivery time can be few weeks.

baudneo, 2023-10-02:
Do you have a paypal?

doctor64, 2023-11-03:
Hi, i finally got bulb from Aliexpress.
Disassemble it and, unfortunately, I was right - it have really different hardware.
Already supported LED controllers get PWM signals directly from Telink TLSR8258 chip and use mosfet for power commutation.
This bulb have SM2335EGH chip connected to Telink chip by I2C and LEDs controlled by it.
Unfortunately, i unable to find datasheet for this chip.
I found implementation for ESPHome/Tasmota https://github.com/esphome/esphome/blob/dev/esphome/components/sm10bit_base/sm10bit_base.cpp and will try to re-implement LED control using it, but it will be not fast.
@Cossid, may be you will be so kind to share datasheet for SM2335EGH?

Cossid, 2023-11-03:
I only originally found parts of the datasheets in Chinese, and originally translated them to English, but they were not perfect, I later found an English version for SM2235EGH, which is better.  That timing and data structures are the same, the only difference is the mA values, but the data representing the table is the same.

 Here are those resources:
 
SM2235 English datasheet:
[SM2235EGH datasheet.pdf](https://github.com/doctor64/tuyaZigbee/files/13254978/SM2235EGH.datasheet.pdf)
Chinese SM2335 power tables:
[SM2335EGH_cn_page7](https://github.com/doctor64/tuyaZigbee/assets/83468485/63b1cf6a-b035-4a91-a52a-2c0d9cad40aa)

Additionally, since then, esp-iot has added drivers as well, for an alternate source: https://github.com/espressif/esp-iot-solution/tree/master/components/led/lightbulb_driver/drivers/sm2x35egh

doctor64, 2023-11-07:
@Cossid Thank you very much for help.
I understand correctly each time when brightness of LEDs changed, so I can determine maximum current of original firmware by capturing any i2c data packet?

Cossid, 2023-11-07:
Correct, you should be able to read the bytes and match them up.  Alternatively, being Tuya, you might be able to read configuration data from their storage key/value area.  They usually store it as `2235ccur` for color current, and `2235wcur` for white current.  It will be the raw byte value, though they sometimes use other names like `ehccur` and `ehwcur` or `iicccur` and `iicwcur`

Cossid, 2023-12-12:
So I bought a random Tuya Zigbee bulb on AliExpress and the model ended up being TS0505B, and it is recognized in Z2M as [TS0505B_1_1](https://www.zigbee2mqtt.io/devices/TS0505B_1_1.html#tuya-ts0505b_1_1).  It appears to be a PWM bulb, there is no LED driver (or even PWM driver) on the LED board, just 6 pins, B-, G-, R-, C-, W-, LV+

The module is [ZTU](https://developer.tuya.com/en/docs/Document/ztu-module-datasheet?id=Kamwy32jy50jp) which Tuya lists as `Z2`, but [the manual](https://fccid.io/2ANDL-ZTU/Users-Manual/Installation-Guide-5037964) clarifies that it is indeed a Telink `TLSR8258F1KAT32`

So be aware, there will be multiple versions with the same base model and will have different pinouts.