doctor64/tuyaZigbee #3: tl_common.h: No such file or directory

MagnusErler, 2023-10-30:
I'm following the guide: https://github.com/doctor64/tuyaZigbee/blob/master/docs/sdk_installation.md
When building the following error appears:
`fatal error: tl_common.h: No such file or directory`
Resource: zb_common.h
Path: /tl_zigbee_sdk/zigbee/common/includes

doctor64, 2023-10-31:
tl_common.h is a part of Telink zigbee sdk, located in tl_zigbee_sdk/proj/
So looks like some misconfiguration in Eclipse configuration.
To be honest, Eclipse is a total mess, i currently working on switch from Eclipse to cmake build.
Please wait few days for update

doctor64, 2023-11-20:
I'm sorry, it takes more time what I expected, but finally all projects converted to cmake build. 
See docs/linux_build.md for details

closed 2023-11-25 by MagnusErler