doctor64/tuyaZigbee #5: Add TYZS3 to support Blind Driver and give Livolo VL-XJ001

sorryusernameisalreadytaken, 2023-11-09:
Unfortunately I am not the only one who has bought a zigbee blind driver that supports zigbee but limits it so much that it cannot be used with z2m or zha.
I'm talking about this Livolo VL-XJ001 device here:
https://www.zigbee2mqtt.io/devices/TI0001-cover.html
https://zigbee.blakadder.com/Livolo_TI0001-cover.html

Livolo devices are apparently only working on channel 26 and with 21758D19004B1200 as extended PAN ID.

I already have another Zigbee blind driver, the one from TUYA: https://zigbee.blakadder.com/Moes_AM43-0_45_40-ES-EB.html
This can be wonderfully integrated into z2m.

I opened both devices yesterday to see what was inside and lo and behold:
Livolo TI0001
~has a PIC16LF18456-E/STX microchip installed as a radio module to which a CC2630 board is soldered

TS0601
~TUYA TYZS3 which, as far as I have read correctly, has an EFR32MG13P732F512GM48 installed

This repo deals with the possibility to flash TUYA Zigbee chips with alternative firmwares. I think that's brilliant!
Now I asked myself the question if I could order a TYZS3 somewhere and if I could read the firmware from the TS0601 to flash it to a new TYZS3?

Unfortunately I can't find any instructions on what hardware is needed to dump a Tuya Zigbee.

Would be great if you could help me here. And sorry this issue is a bit offtopic but this is the closest I could find on the subject.

Here pictures in compare:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

doctor64, 2023-11-13:
I'm sorry to disappoint you, but, unfortunately, this project deals with Tuya devices based on Telink chip. 
I know, it should be clearly stated in project description/readme. 
Your devices have different zigbee chips - Texas Instruments CC2630 on first and Silicon Labs EFR32 on second, and they both have little common with Telink TLSR8258. 
Additionally, i can see on board additional microprocessor, left from radio module on last two photos. I expect it is TI's MSP430G2553 chip, and it handling all hardware motor operations, so zigbee chip only works as network co-processor, just receiving commands from network and pass it to second MCU. Without knowing internal data exchange protocol it will require lots of tracing and analyzing to decipher protocol and make own implementation.
All I can recommend is a find somewhere, may be from another cheap Tuya device, another TYZS3 module, dump firmware from existing module, flash it into new and solder it on device replacing existing communication module.

sorryusernameisalreadytaken, 2023-11-14:
Thanks a lot for clarification on this special case @doctor64 
Can you please add this informations to the main readme? I think you are in a better knowledge position :-)

I will close this here but I you have any knowledge about reading the firmware from a tuya blind driver chip (TYZS3) I would be happy to know that. I read a little bit about a swd-programmer I need? Is that true?

closed 2023-11-14 by sorryusernameisalreadytaken