doctor64/tuyaZigbee #8: any Discord or community ?

KipK, 2023-12-09:
Hi

Grats for this incredible work! 
I have a bunch of different Tuya devices I'd love to see some alt firmwares.

Is there any discord or anything else where we can chat? I'd like to try some implementation but perhaps I'll need a bit of help. 

First I have some TS0052 1ch and 2ch dimmers, their firmware is buggy as hell and having some standard clusters on it would be rzeally neat.

I also have some  other devices to play with if needed:

- TS0202 ( pir sensor ) 
- TS0201 ( Temp / Hum sensor )
- TS0205 ( smoke sensor ) 
- TS110F ( dimmer ) 
- TS0042 ( 2 buttons remote ) 
- TS0003 ( switch module)

doctor64, 2023-12-09:
Hi!
I created Discord channels [#general](https://discord.gg/xSRjUS7Vpy) and [#development](https://discord.gg/GThy6Ednx7) there you can contact me.

About additional devices - to be honest, currently I have around 10 devices waiting in queue for new firmwares.
If you wish to have support for additional devices, please, check what device is Telink based, see some details in [readme.md](https://github.com/doctor64/tuyaZigbee/blob/master/README.md) and create separate issues - only for each device. Please add link where device can by purchased (Aliexpress link or something) and, if it possible, add photos of board.

KipK, 2023-12-10:
Thanks for the links. I wasn't expecting you to do the port, I want to try. But I need to get some spare devices first.

doctor64, 2023-12-10:
Thank you for list. Temp/hum sensor already in my todo, PIR sensor was in work by @suxsem here https://github.com/Suxsem/TS0202-PIR-Sensor-Zigbee-firmware
I hope when we have one of devices implemented, port to other similar devices should be much easier, ideally just provide include file with hardware configuration

Suxsem, 2023-12-10:
Great idea the discord server!
My firmware about the pir sensor is running rock solid since almost a month now and battery voltage hasn't dropped any percentage yet to I'm pretty happy with it.

I was searching a dimmer with the following characteristics:
- generic module, to be connected to a standard wall momentary switch
- mosfet based (instead of triac, in order to smoothly dimmer led lights)
- fade in / fade out controlled indipendently and customizable (I want a 0s fade in effect and a 0.5s fade out effect)
- possible to set the dimming percentage by long pressing the wall switch, with customize speed.

Do you know any zigbee dimmer capable of that? Thank you!

KipK, 2023-12-10:
I saw @suxsem work, will give a try :)

Not sure about the mosfet. All tuya dimmer modules i got have a momentary switch input and react to long press.
All customisation, fade effect and co can be implemented on custom fw. Tuya fw really sucks for this

Suxsem, 2023-12-11:
@KipK I'm not so sure about the effects, because the tuya dimmer I bought has two MCU, one for low level functions (dimming and wall switch input) and one for wireless communication. From what I saw effects are implemented in the first one and I don't think we will be able to write a custom firmware for it

doctor64, 2023-12-11:
@Suxsem 
> fade in / fade out controlled indipendently and customizable (I want a 0s fade in effect and a 0.5s fade out effect)
Actually, if device works accordingly to zigbee standards, any device should can do this. This is called transition and supported by z2m. Or you wish to have transition when you press button/switch on device?

Suxsem, 2023-12-11:
> Or you wish to have transition when you press button/switch on device?
exactly...