doctor64/tuyaZigbee #9: Restoring stock firmware and other questions

ferbulous, 2023-12-10:
Hi, I have these wall switches on Z3L and  TLSR8258

[image]
[image]

1) Can I restore to the stock firmware if I made a backup?
2) Is it possible to get the gpio pinout from the backup firmware so I can configure the pins for relays/buttons on the custom firmware?
3) Can I enable additional feature like decoupling relay and multiple click actions (long/2x/3x clicks)
4) I already have a CH340G, or do I need to use esp8266/esp32 devboard for flashing?

doctor64, 2023-12-10:
First switch, unfortunately, cannot be supported by this project. ZS3L module is EFR32 based, see https://developer.tuya.com/en/docs/iot/zs3l?id=K97r37j19f496
Second yes, theoretically can be supported.
1 Yes, if you make full flash backup before flashing new firmware you can restore to original state. Art least, this works for all devices I worked so far.
2. Unfortunately, no. Or, at least, I don't know how to get this information from tuya firmware. My development process include tracing connections on board to find correct gpio pins. You can see some notes in docs folder
3. Multiclick support in todo list. Relays support need to be implemented, but currently i don't have such devices. If you can provide link to Aliexpress or other store where I can buy your device i can try.
4. I not have success with flashing telink chips using serial port, I use hand made programmer from TB-03 or TB-04 board. See [docs/flash.md](https://github.com/doctor64/tuyaZigbee/blob/master/docs/flash.md)

ferbulous, 2023-12-11:
> I not have success with flashing telink chips using serial port, I use hand made programmer from TB-03 or TB-04 board. See [docs/flash.md](https://github.com/doctor64/tuyaZigbee/blob/master/docs/flash.md)

Thanks, so it looks i'll need to get one

> Unfortunately, no. Or, at least, I don't know how to get this information from tuya firmware. My development process include tracing connections on board to find correct gpio pins. You can see some notes in docs folder

How would test each pins to figure out which function? 
on esphome/tasmota i could just assign all pins as relays and just toggle all relays to figure out which one is relay/led/buttons.

> Multiclick support in todo list. Relays support need to be implemented, but currently i don't have such devices. If you can provide link to Aliexpress or other store where I can buy your device i can try.

I recommend this [switch](https://www.aliexpress.com/item/1005004836095177.html?spm=a2g0o.productlist.main.115.32372e3dKlsL3D&algo_pvid=e882a863-c819-4832-9cba-dd4e6066c9c4&algo_exp_id=e882a863-c819-4832-9cba-dd4e6066c9c4-57&pdp_npi=4%40dis%21MYR%2155.14%2136.03%21%21%2182.55%21%21%402103205217022884131981663ee98d%2112000036222975676%21sea%21MY%210%21AB&curPageLogUid=k4Go0Vwv79zj), avoid this [moes wall switch](https://www.aliexpress.com/item/1005005178430448.html?spm=a2g0o.detail.1000014.3.79e46Yrd6YrdHo&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40000.326746.0&scm_id=1007.40000.326746.0&scm-url=1007.40000.326746.0&pvid=ef1b4418-b4d5-4568-89bd-b080ba5247b6&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40000.326746.0,pvid:ef1b4418-b4d5-4568-89bd-b080ba5247b6,tpp_buckets:668%232846%238113%23679&pdp_npi=4%40dis%21MYR%21103.23%2144.47%21%21%21154.56%21%21%402101eab017022888416813862e11bd%2112000031989000999%21rec%21MY%21%21AB) since that's the ER32 based

doctor64, 2023-12-11:
I found some photos of swith recommended by you and see additional chip near Tuya module. If you have such switch can you please look to marking on it or make a good readable photo?
[image]
I asking because some tuya devices have two MCU - one works with hardware and second, zigbee just works as communication module. I think this approach used to be able to make switches for different radio modules - ie zigbee, wifi, etc. But because of this it will be difficult to create custom firmware for such devices - need to decipher protocol which used by two MCU to exchange data

ferbulous, 2023-12-12:
Sorry, i think you could skip that one since none of my zigbee devices has tuya mcu.

I ordered from alibaba from the second picture

https://x.alibaba.com/15kdOd?ck=pdp

And lonsonho switches (or similar design) don’t use tuya mcu 

https://a.aliexpress.com/_mKcudRa

daniel-dona, 2024-01-09:
> I found some photos of swith recommended by you and see additional chip near Tuya module. If you have such switch can you please look to marking on it or make a good readable photo? [...]

Maybe it's the same Tuya MCU protocol used by ESPHome: https://esphome.io/components/tuya.html

https://developer.tuya.com/en/docs/iot/tuya-cloud-universal-serial-port-access-protocol?id=K9hhi0xxtn9cb

ouinouin, 2025-01-07:
@daniel-dona : the tuya mcu is documented in some opensource projects including esphome and tasmota , there is also some doc available on the tuya website.

doctor64, 2025-01-15:
@daniel-dona  @ouinouin Here is a project working with 2 processor devices of Tuya. https://github.com/slacky1965/tuya_thermostat_zrd Unfortunately, while this allows for fixing some serious bugs in the Zigbee part, it does not enable changing the device's logic, as it is located in the second MCU.