doctor64/tuyaZigbee #11: TS0041: Net Key Button

MagnusErler, 2023-12-27:
I have written the [programmer](https://github.com/pvvx/TLSRPGM) and have tested the connection - all works fine.
I then tried to write the `tuyaSwitch-1button.bin` to my single-chip ZT3L following the [connection-diagram](https://github.com/doctor64/tuyaZigbee/blob/master/docs/flash.md#connection) - write success!
Now, I wonder how to set the chip in pairing-mode. As written in the README I should `Long press on Net Key button` - where is Net Key button?
In Home Assistant the chip was automatically discovered when powering it on. It however was not recognized as TS0041 but as TS0505B (original firmware).

doctor64, 2023-12-27:
Unfortunately, Tuya model names (like TS0041) is more like device class and can't be used to identify different devices.
Switch ZBWS01A used in development have separate Net button. If your switch does not have such button you need to create new device configuration file and possible modify code. Currently my code does not support easy configurations of buttons actions, sorry.

MagnusErler, 2023-12-27:
I ordered [these chips from LCSC](https://www.lcsc.com/product-detail/ZigBee-Modules_YTL-ZT3L_C2897596.html) as I thought I could flash them with new firmware. Would that mean that I for now only can use `tuyaSwitch-1button.bin` for the ZBWS01A?

doctor64, 2023-12-27:
Ahh, you use just board, without device? You need to look in my [notes](https://github.com/doctor64/tuyaZigbee/blob/master/docs/switch-lonsonho.txt)
To emulate switch you need to connect one button to B5 and connect it to Vcc (this will be Net button ), second button to D2 and GND (switch button) and led to D4 and Vcc, and? of course, don't forget about current limiting resistors

MagnusErler, 2023-12-27:
I have flashed my chip with the original firmware of ZBWS01A

```
python TlsrPgm.py -w -t 1 -a 20 -s -p/dev/ttyUSB1 we 0x8000 /home/magnus/repo/tuyaZigbee/originalFirmwares/switch_ZBWS01A/switch_orig.bin 
=======================================================
TLSR82xx TlsrPgm version 12.11.23
-------------------------------------------------------
Open /dev/ttyUSB1, 230400 bit/s... ok
PGM: ChipID: 0x5562 (TLSR825x), ver: 0.0.0.1
swdiv 5, addrlen 3, swbuf [5a 00 06 02 00 05], pwr On
SWire bit rate: 0.9600 Mbits/s
=== PreProcess ========================================
Hard reset Ext.MCU 5 ms... ok
Activate 20 ms... ok
CPU PC=0x00000000
CPU Stop... ok
=== Process ===========================================
Inputfile: /home/magnus/repo/tuyaZigbee/originalFirmwares/switch_ZBWS01A/switch_orig.bin
Write Flash data 0x00008000 to 0x00108000...
 Worked Time: 24.282 sec  
```

When disconnecting and powering on the chip it is still seen as a light in HA (I removed power from the chip, removed the Zigbee device in ZHA, turned on searching for a new Zigbee device, and powered the chip again. It is then still seen as a light...) 
[image]

doctor64, 2023-12-27:
Unfortunately, I don't use ZHA. In z2m you need to to restart z2m after remove.
Second possibility - did you check firmware flashed? If flash are locked progremmer can't write to flash but not signal about it, so you need to verify by reading manually

MagnusErler, 2023-12-27:
I have written `tuyaZigbee/originalFirmwares/switch_ZBWS01A/switch_orig.bin` to the chip, and then read from it again. When comparing the two bin-files they don't seem alike