doctor64/tuyaZigbee #12: Request: OTA update over Tuya firmware (TS0044)

MagnusErler, 2024-01-02:
I have a TS0044 where only switch 4 works (hardware failure). I have added it to Z2M and successfully added [OTA update over Tuya firmware](https://github.com/doctor64/tuyaZigbee/blob/master/docs/ota_tuya.md) even though it says it only works for TS0041. All I did was change all files and names from "TS0041" to "TS0044" (I might not have been the solution). I get battery updates but presses from switch4 aren't shown. I would think that a different output/switch was chosen for the TS0041 firmware. Could you try making a new OTA file for the TS0044, upload it and I could check if it also works for TS0044. Or maybe simply change the input GPIO for the TS0041-OTA file. I'm not able to do it myself - still working on that.

doctor64, 2024-01-07:
Unfortunately, you cannot easy change gpio pins for button in binary. Need to change board configuration file and recompile.
Please, provide pins where switches are connected on your hardware, or, at least, good photo of board and link to store

MagnusErler, 2024-01-07:
**Input:**
sw1 - D2
sw2 - C2
sw3 - C3
sw4 - B4
**Output:**
LED - C0
[ZT3L](https://developer.tuya.com/en/docs/iot/zt3l-module-datasheet?id=Ka438n1j8nuvu)