doctor64/tuyaZigbee #13: What can I do to help add devices to this project? (Specifically TS0012 & TS0002)

roderik-maker, 2024-01-18:
Great project here, really glad I found it. I didn't expect it would be possible to install custom firmware on Tuya devices. What can I do to help add support for the TS0012 & TS0002?

I have 5 modules of the TS0012 and 5 on the way of the TS0002. I would like to bind these devices to various smart lights (primarily TRADFRI & LIVARNO) and stumbled across this project. I don't mind sacrificing 1 or 2 modules (of both types) to this project if this helps to develop firmware that supports binding (and maybe a decoupled mode? [There seems to be some demand for that](https://community.home-assistant.io/t/zigbee-light-switch-with-decoupled-mode/403680/12) ) I'm hoping a lot of the firmware for the TS0041 can be reused for this project?

I've played around with ESPHome for a while and a little with the Arduino IDE (so not a complete beginner) but am well aware that I still have A LOT to learn about micro controllers (I also suck at soldering so would prefer to avoid that as much as possible haha). Anyway, I would like to contribute and was wondering what I can do to help out? Can I just connect a UART <-> TTL adapter to the ZT3L [using this pinout](https://developer.tuya.com/en/docs/iot/zt3l-module-datasheet?id=Ka438n1j8nuvu) and start recording logs? Is that even useful/helpful?
I'm not asking for a step by step guide (unless it already exists and I missed it haha) but some links/tips/other forms of help or guidance would be really appreciated!

Here are some pictures of the TS0012 PCB:     (I can make and share similar pictures of the TS0002 once they arrive, if desired)
[image]
[image]
[image]
[image]

PS. I get into these projects on a sporadic basis (life gets in the way) so sorry in advance if it takes a while for me to respond.

doctor64, 2024-01-18:
Hello and thank you for interest in my project.
Technically speaking, it is possible, but will be some difficulties :)
First, I need to understand schematics of device, at least of some parts, to understand on which GPIO pins external hardware like buttons.switches, relay, leds (if present) connected.
Second, unfortunately, code for relay-class devices not implemented yet. Once it will be implemented, all devices with similar functionality and hardware can share same code with minimal changes, unfortunately, it need to wait some time.
Third, please, provide links where i can buy such device for development (aliexpress is good)

According to your question about logs - yes, it is completely possible, but need to be very careful with such mains powered devices. Often they not galvanic isolated from mains and you can easily fry your PC, serial port, USB_SERIAL converter etc. I already have programmer fried :) And currently it's not have much sense - i don't see any Tuya firmware with any kind of diagnostics log. It will be useful during debugging of new firmware.

roderik-maker, 2024-01-18:
Thank you for your quick response :)

Point 1. I don't mind grabbing a multimeter and seeing if I can Identify the pins related to s1,s2,o1,o2. There are 8 pins going to the controller, I would imagine those are 3V3,GND,s1,s2,o1,o2, reset button and I think it has a status led, so that would make sense. (Should also be easy to identify). Hopefully they're not using any serial internally but I doubt it. 

Point 2. Not sure what I can contribute to this but if I can help let me know.

Point 3.  Purchase links: 
[TS0012](https://www.aliexpress.com/item/1005006089409372.html?spm=a2g0o.order_list.order_list_main.57.558a18027YKL7E) (The 2gang no Neutral variant)
[TS0001](https://www.aliexpress.com/item/1005004371232755.html?spm=a2g0o.order_list.order_list_main.22.4eb41802roOyFd) (The 2CH variant) Let me just double check that this module does indeed also use the ZT3L processor before you invest any time in making this module work. My order should be here at the end of next week... Hopefully haha

Point 4. Okay, if I understood you correctly, I will wait with trying to connect to serial until you have some basic firmware. 

Anything more I can do short term?

doctor64, 2024-01-19:
1. Yes, this can help. Ideally, i need to know also active level of signal, but just basic info will be good for start.
2. Unfortunately, i'm low on free time for project, but i try my best.
3. Ok, waiting for your report on second variant.
4. Yeah, right now it is not needed

roderik-maker, 2024-02-06:
Came back from vacation and the second variant (TS0001) has been delivered, unfortunately it uses the ZS2S and not the ZT3L so that's a no go I fear. I'll have a look at the pinout of the TS0012 (the one with the ZT3L) tonight.

anishp55, 2024-02-06:
I have a few of the temp/humidity sensors - 0xa4c1381b4aef5473.  they are using the ZTU module.  What are the steps needed to add a new device?

roderik-maker, 2024-02-08:
> the contact bounce theory needs to be tested. I think it's not only that, it's just some software limitation (flaw), poor implementation. From a hardware point of view, judging by the schematic diagram, there are no limitations. This is a terrible bug that has been confirmed by various users. I also created a thread on Z2M and it was confirmed there. As for the popularity of the problem - they were sold tens of thousands of pieces :( on aliexpress alone.

With how popular these devices are I am also really surprised that they havn't bothered to fix this in later models. 

Maybe the perceived added value of keeping consumers in the Tuya walled garden outweighs the extra sales they expect to gain from opening up their devices? But I really think they could sell so many more units if they played a little nicer with other Zigbee modules (support binding?) and facilitated OTA updates to fix bugs like this. Really hoping this project can become a solution to this shortcoming :)

Not really sure how much I can contribute to the project at this point (with my current skillset), but I'll keep an eye on this thread (and repo) to see if I can be useful somehow :)

doctor64, 2024-02-15:
Thank you for your work!
From software point of view i don't see anything which can be source of unable to operate both relays in same time. May be some software problem in Tuya sdk - I recently got it but not yet have a time to look in details? but key detection code looks a little, let's say, _strange_ :)
I'm sorry, I don't have enough time to investigate in details. 
I will order both relays in near time, but unfortunately, can't provide any expected schedule for firmware - i need to write full firmware for relay-class device. May be i can reuse light device firmware, but I need to think.

IgnasGi, 2024-02-18:
I bought a dozen MINI-Z1 **2gang** and **3gang** switches with **ZT2S** modules for home automation. But they proved useless with standard firmware. They appear analogous to the TS0003 and TS0012 modules.
How good that this project is - maybe we will be able to do something!
I will make diagrams of the modules, I will be able to test and adjust.
In my preferences.. single, long and double click recognition. It is possible, I hope?
[](https://www.aliexpress.com/item/1005005748125397.html)

doctor64, 2024-03-07:
Hmm, it is possible, but I see two problems here. First, I'm still need hardware details to know on which pins connected buttons, leds, etc. I don't think what flasing TS0041 will help, chances what hardware configuration for TS0041 match configuration of TS0011/TS0012 is pretty small, especially because TS0041 have unusual switches configuration. Action buttons on TS0041 active low, pressing button connect GPIO pin to GND, but net join button on this device is active high, pressing it connect pin to +3.3. 
So I not recommend this.
Ordered 2 gang no neutral version.

monsterkali666, 2024-09-18:
@doctor64 
Hello
I have a bunch of zslc5 zigbee boards,from led bulbs.
i have a jlink v12
dumped the mainflash.hex
is there any chance to convert them to contact sensor? (i need around 15 door /windows sensors,and i was thinking maybe we can repurpose them)
chip is efr32mg21a020f768
simplicity studio fails everytime i try to build project.
i need help
i will pay
also i will provide what you ask for
also if you need the hex,i will upload it
thank you
waiting for your answer

doctor64, 2025-01-15:
@monsterkali666 Sorry for late replay, just returned from hospital.
Unfortunately, no. At least not with this project. Your devices is based on EFR32, which is complete different chip from Telink.