doctor64/tuyaZigbee #14: Support for zigbee plug TS011F

luisfcorreia, 2024-01-29:
Hi, 
how hard is it to support firmware for these types of plugs?

I've got a few, with and without power metering.

Basic support for on/off is all I'm interested in the moment, will be willing to help develop test and debug.

(also, if this isn't the proper project, let me know)

doctor64, 2024-03-07:
Hi, sorry for late reply, missed this issue.
It is depend of real hardware inside, Tuya model numbers like TS011F completely unreliable.
Please, check what device is Telink based. Join device in z2m and look for IEEE Address column. Telink devices should have address started with 0xa4c138. 0xa4c13892ff2b7e05 for example.
If yes, please, provide me with link where device can be bought, aliexpress or something.
If your device NAS-WR01B it's not needed, I already have one
[image]

benbender, 2024-05-08:
Yes, it should be the same HW – for me in Z2M:

```
Zigbee-Modell       TS011F
Zigbee-Hersteller   _TZ3000_gjnozsaz
IEEE-Addresse       0xa4c13876b0fcfdd6
Netzwerk-Adresse    0xC2A0
Hersteller          TuYa
Modell              [TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1)
```

The sticker inside mine says "Plug-001SPB2" and looks identical.

mrmaximas, 2025-03-27:
@doctor64 hi! i have more than ten of this TL8258 based plugs. I have some ota files for this plugs different versions including latest 1.1.2
What needs to be corrected - power, current and voltage reporting algoritm, as well as the upper limit of voltage protection operation, by default it is 264V, it is desirable to increase it to 275V for example.