doctor64/tuyaZigbee #15: Device opportunity: there's no remote relay controlling zigbee thermostat

petermolnar, 2024-02-05:
I've been looking for, including testing, for a battery powered thermostat that would expose on/off direct binding towards a remote relay. In the Zigbee world, I'm fairly certain, it doesn't exist: none of them exposes the relay state. In Z-Wave land https://heatit.com/product/9437/smart-products/thermostats/heatit-z-temp2-white-ral-9003/1109 exists which can do exactly this. 

I tested TuYa X5H-GB-B and TuYa ZWT07 recently, I believe they could both be candidates for this, and the latter is certainly in your compatibility range. While the firmware sucks, the main change would be the output on/off cluster that exposes the relay state. 

I'm pretty sure there would be a market for pre-flashed thermostats with direct binding to remote relay functionality, there are forum posts everywhere asking the same.

So: how could I help?

doctor64, 2024-02-08:
Hello!
It is good idea, unfortunately I have central heating in my home, so i'm looked into radiator thermostat. 
Unfortunately, all TRW tuya models I checked have two MCU architecture - zigbee chip works as Network Processor, only receiving and sending Zigbee messages. Second MCU communicate with zigbee chip over serial channel and actually perform operations with hardware. So this architecture adds additional tiresome task to understand serial protocol between two MCU's or writing new firmware for second MCU.
As I said, i newer used such wall thermostats and don't know in details how they supposed to work. Looking at pictures of ZWT07 i see it have just a relay output where heater can be connected and you wish to add binding support. Seems like it can be done, if device hardware is not dual MCU as i described.
So, please, can you open device, take good photos of board with readable marking and I can check does it have only zigbee MCU.
If yes it can be done, only problem I see can be with advanced features like scheduling.

petermolnar, 2024-02-15:
Hey, sorry this took longer than I wanted to, here are the photos:
[image]
[image]
[image]
[image]
[image]

You are probably correct, there's another IC. What makes me wonder that it's still possible is that there are seemingly unused pins on the zigbee daughter card: could the relay outputs themselves be wired back to the card?

doctor64, 2024-02-15:
Unfortunately, I can't see any marking on second chip on Avatto ZWT-198, and battery cage covers part of traces, but looks like relay control traces going to this chip. Also looks like this chip controls display. So, i fear, this second MCU control all thermostat operation, and zigbee module just handle network communication.
On second device, T5Z, i don't see second MCU, but suspect it hiding under second sticker with QR code, slightly to right, between zigbee module ZT3L and relay module. Also lots of beard traces goes here, I think second chip here, and also controls all hardware.
Answering to you question - yes, probably you can connect relay to zigbee module, but  this will require lots of development. From my point of view it will be easier to just take some relay module, add a temperature sensor and some simple logic like "if temp below threshold turn on relay, if greater - turn off relay." Or just take some thermal sensor with binding and bind it to relay.
Yes, without fancy display and buttons, but in my personal experience anyway nobody use buttons on my TRW's.