doctor64/tuyaZigbee #23: OTA not operational.

doctor64, 2024-04-10:
Trying to update ZBWS01A switch failed - after successful OTA download using z2m, after device restarts, still have previsous version of firmware.
Additional analysis shows what update works when updating from original Tuya firmware to both ver 1.2 and ver 1.3 firmware, but unable to update from 1.2 to 1.3.
Possible reason - original tuya firmware store received OTA image at address 0x70000. Telink zigbee sdk versions 3.6.8.5 and 3.6.8.6 use address 0x77000.

doctor64, 2024-04-10:
Confirmed.
comm_cfg.h fromm zigbee SDK
```c++
#if (BOOT_LOADER_MODE)
	#define APP_IMAGE_ADDR					0x8000
```
part of drv_nv.h from zigbee SDK 3.8.6.5
```c++
#if(FLASH_CAP_SIZE_1M || FLASH_CAP_SIZE_2M || FLASH_CAP_SIZE_4M)
	#define NV_BASE_ADDRESS				(0xE6000)
#else
	#define	NV_BASE_ADDRESS				(0x6A000)
	#define	NV_BASE_ADDRESS2			(0x7A000)
#endif

#define FLASH_ADDR_OF_APP_FW			APP_IMAGE_ADDR

	//1M   Flash: max size = (0xE6000 - 0x8000) / 2 = 444k
	//512k Flash: max size = (0x6A000 - 0x8000) / 2 = 196k
	#define FLASH_OTA_IMAGE_MAX_SIZE	((NV_BASE_ADDRESS - FLASH_ADDR_OF_APP_FW) / 2)
	//1M   Flash: ota addr = 0x8000 + 444k = 0x77000
	//512k Flash: ota addr = 0x8000 + 196k = 0x39000
	#define FLASH_ADDR_OF_OTA_IMAGE		(FLASH_ADDR_OF_APP_FW + FLASH_OTA_IMAGE_MAX_SIZE)
```
drv_nv.h from tuya zigbee SDK
```c
#define FLASH_ADDR_OF_APP_FW			APP_IMAGE_ADDR

	//1M   Flash: max size = (0xE6000 - 0x8000) / 2 = 416k
	//512k Flash: max size = (0x6A000 - 0x8000) / 2 = 196k
	#define FLASH_OTA_IMAGE_MAX_SIZE	((NV_BASE_ADDRESS - FLASH_ADDR_OF_APP_FW) / 2)
	//1M   Flash: ota addr = 0x8000 + 416k = 0x70000
	//512k Flash: ota addr = 0x8000 + 196k = 0x39000
	#define FLASH_ADDR_OF_OTA_IMAGE		(FLASH_ADDR_OF_APP_FW + FLASH_OTA_IMAGE_MAX_SIZE)
```

doctor64, 2024-04-20:
I see few ways to fix:
1: I can change OTA flash addr in SDK. But, this solution will need to patch SDK sources, lose compatibility with devices with correct boot loader, etc.
2: Make new bootloader. But this will broke possibility in the future to flash back original tuya firmware over OTA, complicate build process and user will need to have programmer to flash bootloader and new firmware for first time.
2.1: (improved variant of 2) I can check boot loader version on firmware start, embed bootloader binary into main firmware, check on start and if bootloader is incorrect - flash new bootloader. As above plus additionally complicate code, and need to find a way to embed binary into main firmware image. Telink OTA does not have a way to update bootloader.

Any ideas/suggestion?

pvvx, 2024-04-24:
Tuya uses firmware with a bootloader.
If the custom firmware [is transferred and removes the "bootloader"](https://github.com/pvvx/ZigbeeTLc/blob/master/src/ext_ota.c), then the OTA file must contain the "bootloader" + "application" to restore the original firmware. The original FW is obtained by the sum of two bin files ("loader" + "application") combined into one.

doctor64, 2024-04-24:
Hmm, no. I checked - original tuya firmware on at least one device, switch ZBWS01A accept new custom firmware, put it to address 0x70000 and tuya bootloader successfully transfer image to 0x8000 and loads. Problem is, as I described, what Tuya firmware and bootloader use 0x70000 as address of OTA buffer, but zigbee SDK uses 0x77000, so next OTA update over custom firmware received firmware, but tuya bootloader cannot find it.
I checked tuya zigbee sdk for telink and found some script which concatenate binaries for bootloader and application. But i can't find a code where it can be used.

pvvx, 2024-04-25:
My situation is more complicated - it requires launching with BLE OTA or Zigbee. These are more than 4 launch address options.
As a result, the loaded code, if the markup is not suitable, is always moved to the address space required for work.
The chip starts code from addresses 0, 0x20000, 0x40000 at the “KNLT” label.

1. Tuya BOOT moves the code to 0x08000.
2. Your program moves to 0x40000 upon startup, erasing the BOOT mark.

Next, the usual OTA works in Zigbee, loading the code at 0 or 0x40000.
The option remains when the OTA carries the BLE code, since when loading at 0x40000, a typical OTA will not start. The BLE code checks that it started from 0x40000. Then it moves to 0 or 0x20000.

doctor64, 2024-04-25:
I'm still not sure about switching to architecture without boot loader. Yes, it's easier, but bootloader option allows more space for firmware on 1M chip variants.
No bootloader architecture limits firmware size to 0x40000 - 256K. Architecture with bootloader allows 0x77000-0x8000 = 0x6F000, 444K on 1M flash chips.
PS: Мы можем перейти на русский если удобнее

pvvx, 2024-04-26:
На сегодня максимальный размер прошивки для батарейного устройства, что удалось создать, не превышает 160292 байт. Это [BLE+Zigbee+LCD](https://github.com/pvvx/BZdevice/tree/master/bin).
[Zigbee роутер с функцией управления (HCI) по соединению BLE](https://github.com/pvvx/Zigbee2BLE) и ретрансляцией термометра Zigbee в рекламу BLE укладывается в 212 килобайт.
Больше всего сказываются ограничения в размере RAM и скорости работы данного CPU.
Остатки объема Flash используются для записи истории измерений. Циклический буфер  истории измерений обычно требует большой части линейного объема Flash. По этой причине желательно сжатая компоновка программного обеспечения включающая буфер OTA и настройки.
Для 1МБ Flash (модуль ZTU) [объем памяти замеров выходит в 51943](https://github.com/pvvx/BLE_THSensor) блоков (в блоке: температура, влажность, напряжение батареи, Unix Timestamp).

Загрузчик (boot) в TLSR825x не добавляет функциональности. XIP умеет работать с разных адресов - позиционировать Flash в нужные адреса для данных и исполняемого кода.
Обычно boot применяют на чипах не умеющих работать с разных адресов Flash.

pvvx, 2024-04-26:
> No bootloader architecture limits firmware size to 0x40000 - 256K. Architecture with bootloader allows 0x77000-0x8000 = 0x6F000, 444K on 1M flash chips.

Иногда Tuya в модули ZTU ставят чипы с 512КБ Flash. По этому рассчитывать на 444K для FW не приходится.

pwd-github, 2024-05-24:
Any news on this?

doctor64, 2024-05-28:
> Any news on this?

I implemented bootloader correctly working with OTA images. It is working, but still require a programmer to update bootloader.
I'm need to find a way to update bootloader from OTA, because Telink Zugbee SDK does not have such functionality.

svnhub, 2024-07-24:
Would it make sense to simply make this a per-device configuration? Different vendors may even use different addresses and it just seems very nice to be able to flash both to and from the original vendor firmware using OTA without having to mess with uploading a bootloader or restoring the original one all the time.

Also the calculations in the comment in tuya sdk is wrong:
`//1M   Flash: max size = (0xE6000 - 0x8000) / 2 = 416k`
This is not 416k but 444k as from telink
Did they forget to change NV_BASE_ADDRESS in the example in the comment? Maybe they use some extra space for the tuya settings...

s0meguy1, 2026-09-02:
For anyone with issues with MOEs lights, I figured out the OTA problem with the custom firmware and have built numerous features into the lights. Well, Claude has, I've just been telling it what to do coding wise. We did brick about 8 lights testing (some may have been because of my soldering). Anyway, I've updated around 30 lights so far OTA. You can find the fork here:
https://github.com/s0meguy1/tuyaZigbee

doctor64, 2026-09-03:
@s0meguy1 First of all, i wish to thank you for your work. Idea to use Tuya configuration for device details is really fantastic, may bw it will be a good approach to create config extractor from firmware image, to be able to have two sets of new firmwares - with statically compiled hardware pins assigned  and to use it in runtime.
But, it's LOT's of changes, i'm only started to review it.
I'm trying to split specific device configuration, to have it independently from code. Will think about how to integrate your finding into it.

s0meguy1, 2026-09-03:
@doctor64 - No problem! You are really the inspiration! I saw your repo a while back, tinkered, gave up, then one day I thought "why not use claude?" So this was almost 100% done by claude, I just told it what to do (that and it went over many weeks and bricked lights later that is - ended up just buying the actual ZT3L chips cheap and using those until I got it correct). 

The code base I don't even look at anymore, I really just fire up a coding CLI and ask the question, so I don't know how helpful I'll be personally to you. But it IS working. I just pushed build 37 to 19 lights for a new "self destruct via button push" light show I'm making for my son. These lights can actually do much, much more than the stock software!

s0meguy1, 2026-09-03:
I plan on seeing if its possible to increase the "queue" size for how many light routines can be stored on the device. The problem with a large show and the "queue" is upload speeds if you need it mid show, which is why I want to expand it. I've been trying to reduce zigbee traffic by putting more on the lights, which is amazing for specific things (and timing, holy moly does that help with things) but its a tough balance with some things. Anyway, haven't gotten around it it, I'll keep updating my branch

doctor64, 2026-09-03:
@s0meguy1 Can you please provide some identification info on your lamp - ideally link to Aliexpress, so I can have something to test?

s0meguy1, 2026-09-03:
Absolutely, I actually made a few gifs of some videos I took during bench testing and pushed to git. Feel free to use. Links are there as well:

https://github.com/s0meguy1/MOES-Tuya-firmware#what-it-looks-like

pvvx, 2026-09-05:
1. Is there any point in keeping the Tuya bootloader? As of now, there are already more than five different firmware entry point addresses across various bootloaders. Furthermore, the run entry point address can vary depending on the firmware version within the same device.

This problem is resolved by transferring the [alternative firmware](https://github.com/pvvx/ZigbeeTLc/blob/master/src/main.c#L170) to the required starting address upon initial startup and deleting the bootloader.

2. https://www.tuyaos.com/viewtopic.php?t=5118
_Tuya Zigbee OTA now employs encryption:_
_Starting with TuyaOS version 3.12.0, the TLSR8258 chip platform supports a secure Over-the-Air (OTA) update function._

[image]

This means that for new devices, Zigbee OTA updates are not possible without the encryption keys.

doctor64, 2026-09-05:
@pvvx I'm thinking about adding a bootloader binary built from the SDK to the firmware image  and updating it after OTA. At least for 1M chip.

s0meguy1, 2026-09-05:
@doctor64 One thing I should have mentioned earlier, I think the original problem in this issue is already fixed in my fork. Usual caveat, Claude did all of this, not me. I had it double check the numbers before posting.

The 0x70000 vs 0x77000 thing: it didn't patch the bootloader or change where it looks, it just moved NV. Apparently the SDK computes the OTA bank as `APP + (NV_BASE - APP)/2`, so the default NV_BASE of 0xE6000 lands on 0x77000, exactly what you found. Setting it to 0xD8000 gives 0x70000 instead, which is where the stock Tuya bootloader actually reads from. So the staged image ends up where the bootloader looks and the lights update over and over. I'm ~30 lights in, currently on build 40, reflashing them repeatedly with no programmer.

It's one line in `light/CMakeLists.txt`. It also says 0xD8000 keeps NV fromrunning into the factory blocks at 0xF8000/0xFB000, which the default doesn't.

One thing it flagged that might be worth knowing: NV_BASE and the OTA bank are coupled. Change one and you silently move the other.

pvvx, 2026-09-07:
> [@pvvx](https://github.com/pvvx) I'm thinking about adding a bootloader binary built from the SDK to the firmware image and updating it after OTA. At least for 1M chip.

What is the purpose of the bootloader for an alternative firmware?
Do you use the Tuya IDE and SDK?

Restoring the original bootloader and FW-OTA unit Tuya doesn't always result in a working version. The registration signatures must match...

Or have you decided to work with each firmware version in each variant, which may use different flash memory sizes in TLSR825x/8656 and device configuration in devices with identical names/ID from Tuya, SonnOff, etc.?

pvvx, 2026-09-07:
> I'm ~30 lights in, currently on build 40, reflashing them repeatedly with no programmer.

This doesn't guarantee that tomorrow you'll buy identical-looking tubes, but with different firmware and possibly even a different chip number—TLXR825x or TLSR8656, or with 512 KB of flash memory. Or with a different PCB version—with a different pinout to the main chip...
To restore the original version, in 90% of cases, you'll need to write a full flash dump from the original. This can't be done via OTA.

pvvx, 2026-09-07:
For the TLSR825x and similar chips, reverse engineering has already been done for full debugging with four hardware breakpoints, single-step execution, and the output of all CPU registers. I don't plan to write a full debugger. However, other activists have already partially developed a debugger for Visual Code using this data...
Claude can write and debug a program on Telink chips of this series.
And Telink still hasn't implemented breakpoints and other features in its software and programmers...

s0meguy1, 2026-09-17:
@pvvx Appreciate the notes, and the TuyaOS 3.12 encryption point is a good one. I'll add it to the README so nobody buys new units expecting to convert them over the air.

A few of the things you raise are already handled in the fork, so worth stating them:

- It reads the pin map, PWM frequency and brightness floor from Tuya's factory config block at boot instead of compiling them in. A different PCB revision with a different pinout still works.
- Inherited stock NV is checked at startup and erased before the stack touches it. That was one of the bricks early on, so it's covered.
- The docs say to take a full SWire dump before converting. Agreed that restoring stock is a dump-and-write job, not an OTA.
- Keeping the bootloader is deliberate. This fork targets one product (TS0505B on the ZT3L), and staying compatible with the stock bootloader is what makes OTA work both directions without a programmer. The NV_BASE change just puts the staged image where that bootloader already looks. If a different batch ships a different bootloader, that's a new device to me, not something I'd try to paper over.
- Keeping the IEEE isn't sentiment. 49 fixtures in this house keep their z2m entries, groups, scenes and HA history across the conversion. For a home fleet that matters more than a clean slate.

Your point about calibration data being untrusted is fair. A sanity check on that area at boot is a reasonable hardening; I'll put it on the list.

Usual caveat: Claude does the code, I do the testing on real ceilings. Whatever direction @doctor64 wants to take upstream is his call.