doctor64/tuyaZigbee #24: [Support Request] TS0205_smoke_2 ( smoke sensor )

sorryusernameisalreadytaken, 2024-04-11:
**Z2M Link:** https://www.zigbee2mqtt.io/devices/TS0205_smoke_2.html
**Aliexpress:** https://aliexpress.com/item/1005005863519099.html
**Telink zigbee chip:** [X]
**Model chip:** ZTU
**P/N:** A21-243-YG50A
**S/N:** 10013246700568
**Visible 2nd MCU on board:** [NO]
**Possibly model identification on the PCB:** YG500A A005
**Hardware programmer available:**  [NO] (unsure which one I need)
**Baterry operated:** [X] (2x AAA 1.5V)

I have been watching this repo for a long time and with the TuYa TS0205_smoke_2 Smoke sensor I finally have a device where it might make sense to reprogram the Telink chip.

The device is very inexpensive (approx. 9€), but in my opinion it is missing 4 important features:
- switch off alarm via zigbee (may not even be technically possible for hardware)
- test/activate alarm via zigbee (may not be technically possible)
- alarm silent mode (may not be technically possible for the hardware) (for the purpose of using the fire detectors additionally)
- put neighboring TS0205_smoke_2 also in alarm mode (may not be technically possible for hardware) (is telink able to use zigbee binding?) (may need additional power)

Pictures:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

doctor64, 2024-04-15:
First of all - great bug template! Can I use it?
About hardware programmer - I recommend make your own SWS programmer from TB-03F or TB-04 boards, as described in docs. Serial ports programmers are, in my experience, unreliable, Telink BDT is too expensive.
Ordered device, waiting for delivery.
About features:
Alarm operation (items 1-3 in list) - I understand you mean alarm sound? Yes, need to understand schematics, but i think it is possible.

> put neighboring TS0205_smoke_2 also in alarm mode
You mean sound alarm or zigbee logical alarm state? 
Any device can be binded, of course, from software point of view binded device just receive command to cluster. But, this can be done on cost of battery life. To ZED (Zigbee end device, usually battery powered) receive any command on zigbee network, it must checks his uplink for incoming zigbee messages. Standard routers timeout for keeping messages for its childs is a 7.5 seconds, this mean what end device need to checks in intervals smaller than this. ZCL standard recommends 5 seconds, and my implementation of POLL_CONTROL cluster support use this value. So, in default configuration bindidng should work, with delays up to 5 seconds.
This timeout configurable by user, so it is possible to select between binding/response time and battery life.
From software point of view all this can be implemented, just additionally to IAS Zone cluster (which is already implemented) we need to implement IAS WD (warning device) cluster.

sorryusernameisalreadytaken, 2024-04-16:
> First of all - great bug template! Can I use it?

For sure.

> Ordered device, waiting for delivery.

Very good. I think it could be a really really cool custom device. I will wait for your reply

> About features: Alarm operation (items 1-3 in list) - I understand you mean alarm sound? Yes, need to understand schematics, but i think it is possible.

I guess yes :D

> You mean sound alarm or zigbee logical alarm state?
My basic idea was to set other [TS0205_smoke_2 ( smoke sensor )] also in a mode wear a possible fire victim can hear it. E.g. there is a fire in the living rom, the acustic in bethroom starts too.

doctor64, 2024-04-17:
> My basic idea was to set other [TS0205_smoke_2 ( smoke sensor )] also in a mode wear a possible fire victim can hear it.

I understand. ZCL standard cover this, but it expected what all this tasks performed by some central device in network, CIE. See pic below. Unfortunately, Z2M implement just minimal subset of CIE functionality.
[image]

I can achieve what you wish but:
1. Devices need to poll network at least every 7.5 seconds (hoping what routers in network conform to standard) 
2. You cannot directly bind IAS Zone cluster to IAS WD (warning device) cluster, because, well, it's different clusters. I can add genOnOff clusters to both sensor and warning parts, so you can bind any device with OnOff cluster (switch, button, my door sensor, smoke sensor, etc) to your siren and sound it. But this solution will not have any real security. Anybody who knows your network key can start siren or stop it

doctor64, 2024-04-30:
Device arrived, in development queue

luckylinux, 2024-11-29:
First of All, Great Project :+1: 

I'd also be interested in this, if nothing else to have a further Firmware Option, since there have been Reports of frequent Firmware Disconnections (apparently Fixed in one of the latest zigbee2mqtt Releases).

Do you have a Status Update or an Idea when this could get Implemented ?

Thanks :smiley:

ckglobalroaming, 2024-12-19:
I have many of these ZTU smoke detectors and they are very bad zigbee devices. Anything that comes from this is greatly appreciated. Thanks

doctor64, 2025-01-15:
@luckylinux @ckglobalroaming Sorry for late replay, just returned from hospital. Fear not, project not stopped, just a little suspended. I promise to continue working on it, unfortunately can't provide any reliable estimations now.