doctor64/tuyaZigbee #28: [Support Request]Zemismart ZM-H7

dekamilv, 2024-04-18:
[image]

Hi Alex. 
May I ask you to add support of very nice and beautiful switch Zemismart  ZM-H7.
Problem of all thees switches in unexpected ON/OFF time to time.

Device based on ZYZBP012 chip.

[image]

[image]

Chip itself only communicate with the network and everything else controlling by 2x mcu. 
I was lucky to figure out controlling pins. 

[image]

B5 - "button" - quick changing on off/on signal (high/low) when proximity sensor activated
C3 - relay state 
C2 - responsible for white colored light but no ideas why thats implemented - it is not using by switch anyway and moreover broke illumination logic if switched on for more then 3 sec.   

"On plate" button is responsible for pairing and makes long "on" signal (while pressed) on B5.

Any chance pls ?

dekamilv, 2024-04-19:
> Interesting device. As I can see, it uses mm-wave radar HLK-LD101 for hand detection. https://www.hlktech.net/index.php?id=968 If this radar module does not require some specific configuration, this device can be supported - but only then I have finish relay class device implementation. But, i fear that this random unexpected ON/OFF is caused by false detection or interference with radar and cannot be fixed by new firmware, at least without changing configuration of radar. Can you please check what two small IC (second photo of blue board, right side) have any markings?

dekamilv, 2024-04-19:
Many thanks for your attendance. 

This switch is really working nice in comparing with optical models - bearing in mind self switching of course. 

I am sure module doesn't communicate neither with mm-wave module nor with IC (by thу way - no any marking on chips present).
Was lucky to buy similar model with WiFi chip.

[image]

As it turned out WiFI model is based on absolutely same board platform with identical number YC-ZM-SS-01-A-V2.3 and differ only by presence of WB3S communication chip instead of Zigbee. TX / RX / CEN pins "floating in air" and it was really easy to download TUYA firmware to get control pins and then install esphome to let it work. 

Also I pretty sure random unexpected occasions not caused by radar-module interference:
1 - because happens only unexpected ON-state and then never OFF - sorry, I did not describe it clearly in the first post.
2 - WiFI model working already a week on a same place where zigbee model were installed. No any unexpected actions registered.

dekamilv, 2024-04-19:
Sorry, have no clue why topic closed....

doctor64, 2024-04-20:
Ok, thank you for updated info. 
Can you provide link to Aliexpress or verify what this is correct device? https://www.aliexpress.com/item/1005005175284937.html

dekamilv, 2024-04-22:
> Ok, thank you for updated info. Can you provide link to Aliexpress or verify what this is correct device? https://www.aliexpress.com/item/1005005175284937.html

Yes. I bought exactly from this link :)

doctor64, 2024-04-30:
Device ordered, please wait for delivery and development