doctor64/tuyaZigbee #32: New Device Request: MOES TS0011 /TS0012 / TS0013 light switches

CRCinAU, 2024-04-26:
Devices as discovered in Home Assistant using ZHA:

TS0011 - _TZ3000_hhiodade - 1 gang
TS0012 - _TZ3000_18ejxno0 - 2 gang
TS0013 - _TZ3000_qewo8dlz - 3 gang

Pictures of the TS0012:
[image]
[image]

The 8 pin black connector goes to a rear board with relays etc. I'm guessing that's not really important at this point.

Happy to do any probing that is reasonable to see what I can figure out...

CRCinAU, 2024-04-26:
The switch configurations seem to show:

TS0011, R41 is populated, R40 and R42 are not
TS0012, As shown, R40 & R42 populated, R41 not.
TS0013, R40, R41 and R42 are populated.

There is a TS0014 - that's a 4 gang version of this - but I believe it'd have to be a different PCB - as the switches are arranged in a square - not all in-line. I don't have one of these.

doctor64, 2024-04-30:
Please provide device name, Tuya names like TS0011 are unreliable and provide link to shop, Aliexpress preferred

CRCinAU, 2024-04-30:
The device names are the second field in the first post - ie `_TZ3000_hhiodade`

The device manufacturer is here: https://moeshouse.com/

AliExpress Store: https://www.aliexpress.com/store/1102476055

CRCinAU, 2024-05-03:
So, as a quick review of these units....

1) They do work with no neutral connected.

2) If output L1 is connected to a low wattage LED light, the LED light may flicker when on. I'm waiting some capacitors to hook between L1 and neutral that MOES support say should resolve this. Additional down side, they don't supply these caps.

3) It doesn't seem like the firmware supports binding to groups (or I couldn't get it working in ZHA) - meaning all groupings etc need to be done via HA automations.

4) The current firmware only supports ENDPOINT mode - so will not work as mesh / router devices. Likely this was done for a no-neutral install case - however even with a neutral wire connected to the device, it only works in endpoint mode. I really wanted to extend my mesh with these - being hard wired - but this is not currently supported.

5) When installing, the touch pads have a plastic film on them. I'm not sure if this should be taken off to make the touch more sensitive. I can't find anything written about this.

6) The LED on the switch has the option to set a couple of different modes. Home Assistant seems to have these as a "TuyaZBOnOffAttributeCluster" - whatever that is - as "backlight_mode". I use SwitchBackLight.Mode_2 - which is the LED on when the switch is off, LED off when the switch is on.

Mostly, they "just work" in Home Assistant - but it would be nice if groups worked properly, and they could be used as routers...

MnM001, 2024-05-03:
Hi,

I have these devices too but they look a bit different:

[https://access.remotedpi.online/#/client/MgBjAG15c3Fs](url)

I am using z2m with them and while the names might be different in z2m (like 1 gang are not identified as TS0011 but as Moes ZS-EUB_1gang for example) I believe they are the same kind of devices. I also have the 4 gang one:

_TZ3000_hhiodade - 1 gang - identified in z2m as Moes ZS-EUB_1gang
_TZ3000_18ejxno0 - 2 gang - identified in z2m as Moes ZS-EUB_2gang
_TZ3000_qewo8dlz - 3 gang - identified in z2m as TuYa TS0013
_TZ3000_mrduubod - 4 gang - identified in z2m as TuYa TS0014

It will be so good if these devices will be supported. I understand that initially there will be work done to allow moving away fro Tuya, however I wonder if the following will be supported as well:

- detached mode
- bindings
- maybe make them work as routers (currently hey ony work as end devices)
- button led support (have the led on or off
- power-on behavior (previous, off, on)

Please let me know if I can help with anything - sadly I am not good with coding or writing software. I can help with testing, information required, etc

You are doing an excellent job!!!

CRCinAU, 2024-05-04:
> _TZ3000_mrduubod - 4 gang - identified in z2m as TuYa TS0014

Are you able to do pictures of the main PCB on this one? I have the 1 / 2 / 3 gang versions, but the 4 gang seems to be a completely different PCB layout. More thinking of documenting the differences at this point...

doctor64, 2024-05-04:
@CRCinAU 
1 and 2 - I fear nothing can be done on hardware part from my side.
3 Tuya devices rarely supports binding. I implement this.
4 I'm really not sure what can be done here. No neutral devices consume power from passing current with device load, which can cause LED lamps blink, so all implementation I have see so far try to minimize power utilization. It's why such device not support route function. Anyway, it will require some experimentation to see how device will behave with routing enabled. Unfortunately, routing cannot be switched on/off on the fly, it is determined during compilation.
5 no info here :)
6 I think it can be determined later. Currently, per channel LED just blink on key press so user can get some feedback. 

Question from my side - this devices don't have dedicated Net key for device join, right? How network join initiated on original firmware?

doctor64, 2024-05-04:
@MnM001 
please, drop all characters on aliexpress links after .html - it is not needed and confused github :)
Use only https://www.aliexpress.com/item/1005005178075186.html
About feature lists:
Detached mode - planned, may be in not first release.
Binding - definitely, i working on two way binding - so you can bind some other relay to button on your wall switch and separately bind relay part of switch to some other button/switch
Router - see previous answer. Need more experiments.
Led control - again, planned, but not in first place. I'm already have too many features and too little time for all it :)
Power on behavior - definitely, it's already implemented for lights devices

Please, work with @CRCinAU  to determine if this two devices actually the same from hardware point of view. Tuya names can be different for completely identical devices, and, unfortunately, identical devices can have different hardware. Firmware need to match hardware. If devices are different, then yes, I need to have separate issues.

CRCinAU, 2024-05-05:
> @CRCinAU 1 and 2 - I fear nothing can be done on hardware part from my side.

Yeah - I understand this - it was more a review of the device in general :)

> 3 Tuya devices rarely supports binding. I implement this.

Same as above. It's a limitation of the factory firmware.

> 4 I'm really not sure what can be done here. No neutral devices consume power from passing current with device load, which can cause LED lamps blink, so all implementation I have see so far try to [...]

Yeah - this is what I was wondering. In my install case, I *do* have a neutral wire connected to the device. It has an 'N' wire connector, and looking at the rear PCB, it does look like its connected to something... I followed the track on the PCB for a bit, but couldn't tell if its properly utilised or not.

> 5 no info here :)

Again, more of a general review of the device :)

> 6 I think it can be determined later. Currently, per channel LED just blink on key press so user can get some feedback.

Yeah - having the LED turn ON when the switch is OFF is great for finding them in the dark :)

> Question from my side - this devices don't have dedicated Net key for device join, right? How network join initiated on original firmware?

This seems to be the same as most devices - if you hold any of the switches for ~10 seconds, LED1 will start to blink and the devices goes into its join mode.

MnM001, 2024-05-05:
> > _TZ3000_mrduubod - 4 gang - identified in z2m as TuYa TS0014
> 
> Are you able to do pictures of the main PCB on this one? I have the 1 / 2 / 3 gang versions, but the 4 gang seems to be a completely different PCB layout. More thinking of documenting the differences at this point...

I have them all installed and is a bit hard to take it out.
All I have a  2 button one as spare - I will take some pictures to compare to yours and see if they are the same inside.
I guess initially we can focus on the 1, 2 and 3 button ones. If we can get them going that will be excellent.
Then later on we can revisit the 4 button one and work on that.

benbender, 2024-05-08:
I'm also having several of https://www.zigbee2mqtt.io/devices/ZS-EUB_1gang.html#moes-zs-eub_1gang and [ZS-EUB_2gang](https://www.zigbee2mqtt.io/devices/ZS-EUB_2gang.html#moes-zs-eub_2gang) installed here and would be very interested in an open firmware with extended capabilities – especially in regards of binding. If there is anything you need or I could help with, please let me know!

CRCinAU, 2024-06-21:
@doctor64 - Just wondering if your device arrived, and if you've had a chance to play with this as yet.

MnM001, 2024-10-04:
Hi @doctor64  have you managed to make any progress on these devices?
Anything we can help with?

uniquePWD, 2025-01-17:
Get well soon @doctor64 

We truly appreciate everything you're doing as part of this project, but I appreciate your health more

CRCinAU, 2025-01-17:
@doctor64 Yeah dude - make sure you take care of yourself as a #1 priority...

MnM001, 2025-01-31:
@doctor64 - Nice to see you back and sorry to hear about your health issues.

I hope your recovery is going well. Take you time and make sure you are getting better. Health is priority number one! Our ZigBee devices can wait :)

romant17, 2025-02-04:
> > _TZ3000_mrduubod - 4 gang - identified in z2m as TuYa TS0014
> 
> Are you able to do pictures of the main PCB on this one? I have the 1 / 2 / 3 gang versions, but the 4 gang seems to be a completely different PCB layout. More thinking of documenting the differences at this point.

Not sure if this is still relevant but my 4 gang device identifies as Zigbee Model TS0014 Zigbee Manufacturer _TZ3000_mrduubod

[image]
[image]
[image]
[image]

I have a problem with their behaviour in z2m - the switches operate manually and can be switched by HA as well but the status of the  switch in HA does not update when the switch is operated manually.

A standard mqtt payload looks like this 

`{
  "backlight_mode": "low",
  "last_seen": "2025-02-04T11:55:17-04:00",
  "linkquality": 94,
  "power_on_behavior": null,
  "power_on_behavior_l1": "previous",
  "state_l1": "ON",
  "state_l2": "OFF",
  "state_l3": "OFF",
  "state_l4": "OFF"
}`

the 4 state_lx pairs properly reflect on/off but this doesnt seem to reach HA. Is there a way to correct this without doing custom template sensors?

MnM001, 2025-03-13:
@CRCinAU @benbender see here guys:

https://github.com/romasku/tuya-zigbee-switch/issues/14

Cheers