doctor64/tuyaZigbee #33: I would be keen for this as well (same use case) including decoupling the buttons and switches.  I have these https://www.zemismart.com/products/kes-606us-l1 which come with a ZS3L chip (also on aliexpress here:https://www.aliexpress.com/item/1005005396473245.html).  I have done a little bit of investigation, and looks like the input switches are connected to pins 2,3,4,5 and the output relays connected from pins 10,11,12,13 (this is from quick investigation. need to buzz out all the connections to 100% confirm).  Photos of inside the unit here

doctor64, 2024-04-30:
I would be keen for this as well (same use case) including decoupling the buttons and switches.  I have these https://www.zemismart.com/products/kes-606us-l1 which come with a ZS3L chip (also on aliexpress here:https://www.aliexpress.com/item/1005005396473245.html).  I have done a little bit of investigation, and looks like the input switches are connected to pins 2,3,4,5 and the output relays connected from pins 10,11,12,13 (this is from quick investigation. need to buzz out all the connections to 100% confirm).  Photos of inside the unit here
[image]
[image]
The relays are connected to the 8pin Header on a separate PCB (along with the powersupply etc)

I can help and test if needed, just need a few pointers, as not a very strong programmer.

_Originally posted by @jbro164 in https://github.com/doctor64/tuyaZigbee/discussions/31#discussioncomment-9261565_

jbro164, 2024-05-02:
I have traced through the wiring, and here is the confirmed allocated pinout for the device
[image]

The PCB is the same for 1-4 (looks like it can do 5 as well, but this is not a purchasable product), so for the other versions, will just be deactivating the circuits.

The Inputs (Buttons) are Active Low (i.e. internal Pull-ups enabled as there are no external pull-up resistors).

The Outputs are active high (i.e. Relay closed when pin is high)

doctor64, 2024-05-02:
Thank you for info.
Just a question - I don't see separate button for network join. How device with original firmware joins to network?

jbro164, 2024-05-02:
It was already in pairing mode when I received it, so way easy.  From the instructions it says to "Hold any one switch button for 10-15s until the indicator lights flash quickly"  When placed in pairing mode, all LEDs flash around 2x flashes per second (equal on and off times)

jbro164, 2024-05-02:
Also realised that when the LEDs flash, the output contacts didn't operate (which are from the same pin).  Discovered that the negative of the LEDs is via a transistor connected to the top right Pin, and this is how they are flashed  The pin is normally high. and drops low when needed to stop the LEDs (Note that the LED status is opposite to the relay outputs, i.e. off when the relays are on, so only the LEDs that have their relay off end up flashing, the others just stay off)

doctor64, 2024-05-04:
@jbro164 
Good finding and really bad news ;(
As you understand, i cannot have separate code for each device, so i try to write generic code and differ only in hardware details. But this strange approach to led control make it difficult to implement in generic form, because current implementation expects that each external (led, relay, button) connected to dedicated gpio pin.
Can you, please, check, if relays also have similar enable pin?

jbro164, 2024-05-04:
No, there is not a similar one for the relays.  that is clear as they are connected through pins to the relay boars which are all labelled (and i have checked the spare pins are all spare)

doctor64, 2024-05-05:
Hmm, i need really think about this. May be we can think what device have only one logical led (this control pin for all leds), and just operate on this for status information (joining, etc).