doctor64/tuyaZigbee #36: [Support request] Zigbee-01-5CH

FaBjE, 2024-08-12:
I own a series of [Zigbee-01-5CH](https://nl.aliexpress.com/item/1005006257020705.html) LED Strip controllers.
As far as I can see they are pretty similar (maybe even identical) to the [already supported LED controllers](https://github.com/doctor64/tuyaZigbee/blob/master/docs/devices.md#schematics).

[image]

The only difference being that these do not have an LED or button to join the ZigBee network.
So supporting these would require the "Reset by turning power on/off 5 times" to be implemented.

Although the current tuya firmware is "ok", it is not perfect.
For me for example the lowest-dim-level is still to bright. the "Direct control of LED PWM" (as on the todo list) would be awesome!
Save-state and power-up control would be a great benefit as well.

Let me know what I can do to help.

serpent213, 2025-03-10:
For the engines:  AP-Zigbee-01-DIM, AP-Zigbee-01-2CH, AP-Zigbee-01-3CH, AP-Zigbee-01-4CH, AP-Zigbee-01-5CH

[image]
(slightly different silkscreen?)

serpent213, 2025-03-10:
> Although the current tuya firmware is "ok", it is not perfect.
> For me for example the lowest-dim-level is still to bright. the "Direct control of LED PWM" (as on the todo list) would be awesome!
> Save-state and power-up control would be a great benefit as well.

@FaBjE, sorry for being off-topic, but info on these modules seems to be sparse, and I'm fairly new to Zigbee and Tuya.

Is there a way to pair these with a generic Zigbee coordinator (zigbee2mqtt) using the stock firmware? Couldn't figure it out.

thekief, 2026-01-11:
@doctor64 @FaBjE did you look any further into this then?