doctor64/tuyaZigbee #37: TH02Z Temperature humidity sensor

auron-muc, 2024-08-30:
I saw the TH02Z Temperature humidity sensor is already in the WIP list, and there is also [tempSensor-notes.txt](https://github.com/doctor64/tuyaZigbee/blob/master/docs/tempSensor-notes.txt) with all the info regarding the module. 

I've got a few of them, and would like to replace the firmware because the stock FW is very inconveniant to set up and kind of unreliable with HomeAssistant.

Is there anything I could do to help here? I've done some ebedded coding, but no zigbee yet. I would need some pointers where/how to best start.

I already got the programmer and extracted the stock FW from several devices. 
They use the ZTU board and a PCB with the marking "RH-MAGv3B".
No need to solder for me, all pins needed for programming are very conveniently on the ISP header:
[image]
Different from your instructions I needed to connect reset though, the python tool would not work without.

D4 on TB-03F-KIT goes to SWS on the TH02Z 
D3 on TB-03F-KIT goes to RESET on the TH02Z 
GND and +3v3 are self-explaining.

auron-muc, 2024-08-30:
button goes to pullup restistor R6  and then to pin_3/C0 on the ZTU
LED goes B4 on the ZTU -> RS0 -> LED -> GND
CHT8305 :
SDA -> pin_8/C2 on ZTU (pullup R4)
SCL -> pin_9/C3 on ZTU (pullup R3)
Alert -> pin_1/D3 on ZTU

I took an image of the back and front, light intentionally from the side to better see the traces. Front is a bit blurry due tu this "bad" light, but still visible OK.
[image]

baetzst, 2024-09-20:
There is already a custom firmware from pvvx.
https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-d3a3-01993001-TH02Zz_v0123.zigbee