doctor64/tuyaZigbee #38: TS0042 support request.

JokinenK, 2024-09-03:
Hello! 
I bought one of these cheapo 2-gang scene switches and would love to see this getting proper firmware.
I attached two pictures I took of the PCB and under is the pinout I measured with multimeter (continuity / beep).

S1 = D2, LED above = D4
S2 = C2, LED above = C4
S3 = C3, LED above = D7

Is there something else I could do to make it happen? I would love to use it as a dimmer and as for the vanilla firmware it just does not play along.

[image]
[image]

JokinenK, 2024-09-06:
Zigbee2MQTT reports device as `_TZ3000_tzvbimpq`.

CaptInsano, 2025-04-01:
I came looking for the nearly the same thing:

My device is the TS0043 3-gang wireless switch.

PCB (see below) looks the same as above but populated for 3 switches.

[image]

[image]

Zigbee2MQTT reports device as below:

[image]

and links to this page: https://www.zigbee2mqtt.io/devices/TS0043.html#tuya-ts0043

The main difference that I can read between our device and the supported device:  "Lonsonho 3 gang switch" (from list of supported devices on this repo) is that our device does not have a separate pairing button, pairing is initiated by holding down S1 for 10 seconds.

The main reason that I am looking for a custom firmware solution for this device is that the long press function of this switch is terrible, it only reports a long press **after** the button is released, making it useless for automations to dim lights.

Do you think there is any chance support for this device could be achieved with this firmware?