doctor64/tuyaZigbee #43: TS0001/TS0002 support request

alexandrsafronov9, 2025-01-03:
Hi friend.
I purchased EKF devices https://ekfgroup.com/ru/catalog/products/umnyj-vyklyuchatel-stokgolm-1-kl-besprovodnoj-belyj-zigbee-ekf-sonnect and https://ekfgroup.com/ru/catalog/products/umnyj-vyklyuchatel-stokgolm-2-kl-belyj-zigbee-ekf-sonnect
The TUYA ZTU module is installed inside.
Zigbee2MQTT reports device as  _TZ3000_fwp3gou5 /  _TZ3000_t8xva7xy
Is it possible to support these devices? It would be great to make one, two and long press on the key and control the indicator LEDs. 
On the single button switch there is no way to change the behavior of the LED. It lights up when the switch is on and does not light up when it is off. I wanted to make a night light out of it so that the switch could be seen. 
On the double button, you can control the behavior of the LED.
What do I need to provide to make the firmware?
Thank you.

alexandrsafronov9, 2025-01-03:
[image]

doctor64, 2025-01-17:
Sorry for late replay, have some health issues. First device can be supported by existing code, but I need to know hardware details, so ideally I need device for checking. Unfortunately, I'm not in Russia, so buying one will be problematic. May be you can check on AliExpress for such device? Or at least provide good photos of board on both sides. Multiclick support in development right now.

> On the single button switch there is no way to change the behavior of the LED. It lights up when the switch is on and does not light up when it is off. I wanted to make a night light out of it so that the switch could be seen.
As I understand it is a battery powered device? Keep LED on will eat battery quickly.

Second device, as I can see is a relay switch without neutral. Implementation of such devices is next in line. 
Can you check what both devices use ZTU module?

alexandrsafronov9, 2025-01-23:
Hi Alex.
I believe that your health will be fine. Both devices are connected to the break of the live wire without neutral. There are no batteries in the devices. I am sending a photo. Both modules use ZTU.
I can send you both devices by mail.
Thank you.

[image]
[image]
[image]
[image]
[image]

Hunter32R, 2025-01-23:
This firmware will most likely suit you: [https://github.com/romasku/tuya-zigbee-switch](url)

alexandrsafronov9, 2025-01-24:
Hi Alex!
I tried the firmware and it doesn't work for my devices. All hope is on you.
Thank you.

alexandrsafronov9, 2025-03-28:
> Hello alexandrsafronov9,
> I see that you also have the same chip ZTU as the on I just posted #44
> Can I ask you which ports you connected the GND, V3.3 an SWS. Also
> which programer kit did you use. Yor help will be greatly appreciated.

Hi friend.

On the back of my device, all the connection points are written on the 
board.

28.03.2025 14:23, shsurury пишет:

alexandrsafronov9, 2025-03-28:
Hi friend.

On the back of my device, all the connection points are written on the
board.

[image]

shsurury, 2025-03-28:
what firmware worked for you

shsurury, 2025-04-26:
Can you please help further. I connected as per your picturte but when I run python TlsrPgm.py -s -p/dev/ttyUSB0 i get an error 
TLSR82xx TlsrPgm version 31.10.23
-------------------------------------------------------
Open /dev/ttyUSB0, 230400 bit/s... Error!
any help in wiring or sequen of connecting wil be greatly appreciated.